matthijskooijman/arduino-dsmr #2: hanging at ParseResult (Parse example)

rubenverhoef, 2016-04-25:
Hi, 

I was testing the Parse example but my code is haning at ParseResult. 
`  ParseResult<void> res = P1Parser::parse(&data, raw, lengthof(raw), true);`
I have a print before that line and after but i only get the print before that line.

I use the Arduino Nano, have you any idea's?

matthijskooijman, 2016-04-25:
Hm, that's odd. I had a quick look through the code (it's been a while since I wrote it), but couldn't find any obvious culprits for an infinite loop (though there are plenty of loops in there). Perhaps it would be useful to look into the parse function (and the functions it calls) and add some prints there? The code is a bit complex due to the templates, but the general flow should be easy to follow.

Also, make sure to add a `Serial.flush()` after every line, to ensure that all data is sent through serial before continuing (normally, things are queued and sent in the background).

rubenverhoef, 2016-04-26:
Arduino Mega (ATmega 2560) is working, Arduino nano is not, so it has something to do with the ATmega 328 chip...

My two Arduino Nano's are behaving strange, when i add some prints with flush it works kind of, there is data coming but not the same data as on the ATmega 2560 and the data is also repeating. Really strange behavior because I have to set 3 prints, one on the main program and two in the parse file and the place is not really defined, every place gives an other result.

Sorry for the bad description but it is really behaving strange.
(also tried on older arduino version (1.6.6))

matthijskooijman, 2016-05-13:
I finally got around to digging in a bit further. Turns out that the huge list of fields both examples caused a very deep call stack, since the parse function works recursively. Because each call allocated memory for the parse result for that call, this quickly used up a lot of memory, overflowing the stack on the Arduino Uno. 34ab0ff shows that I was actually aware of this limitation, but I apparently failed to document (or remember) it.

I've made some changes to the library to force these recursive calls to be inlined, which makes this  problem go away and lets the examples run on an Arduino Uno as well (though I only tested `parse.ino`, not `read.ino`).

Let me know if the current git masters works for you, then we can close this issue :-)

rubenverhoef, 2016-05-13:
yeah! the parse example is now working on my arduino nano!

Now i will look into the serial reading

closed 2016-05-13 by rubenverhoef