matthijskooijman/arduino-dsmr #5: Parser doesn't always handle constant stream of data well

thomasvnl, 2017-01-03:
Hi,

I've been working with this library for a few days now and I feel a bit stuck. That has probably a lot to do with the way I'm using the library. I created a small little shield to connect to my P1 port on the smart meter. This shield has the request pin always at high. When initializing I just use a dummy pin to set "high" when I call reader.enable() from your library to trigger a read, in conjunction with a SoftwareSerial object. It is starting to look like something isn't flushed totally and because of that, I sometimes get a lot of "OBIS id Empty" messages, because there were multiple DSMR messages being read at once (?). Also I frequently experience "Invalid unit" messages but that looks more like a corrupt message to me (which doesn't bother me that much to be honest).

I've tried using reader.enable(false), removing the millis() check in the loop and still running reader.loop inside the arduino loop but to no avail. 

I was wondering if you've experienced these kind of errors (well duh, you made the library) and if you have any suggestion on how to work around / fix this?

With kind regards,

Thomas

thomasvnl, 2017-01-03:
After ruling out modules, it seems to be the SoftwareSerial implementation I was using has a problem with timing, causing it to frequently supply corrupt messages where for instance the "/" was missing (because it didn't pick up the rising edge fast enough I guess). Switching to the builtin serial port resolved all issues I have so I'm sorry for blaming your library and I'm closing this issue ;).

closed 2017-01-03 by thomasvnl

matthijskooijman, 2017-01-03:
Thanks for the followup, good to hear it's working correctly now :-)

thomasvnl, 2017-01-03:
I do want to thank you for writing such a nice library to parse DSMR data. It's really something! I also just noticed that you live like 1.5km from me, so hello relatively close neighbour.

matthijskooijman, 2017-01-03:
> I do want to thank you for writing such a nice library to parse DSMR data. It's really something!

Thanks. The irony is that I'm not actually using it myself yet, due to some hardware supply issues my home automation project got a bit delayed, but hopefully I'll be wiring up my meter in the next months.

> I also just noticed that you live like 1.5km from me, so hello relatively close neighbour.

Small world! :-)