matthijskooijman/arduino-dsmr #7: Errors in read example

IncognitoCore, 2017-11-03:
I'm getting [a lot of errors](https://www.dropbox.com/s/pjaoo8kip0iw7ne/error_messages.txt?dl=0) when I try to use the read example.

Note that I'm using an Arduino UNO so I'm using Software Serial.

This is what I have edited in the file:
#include <SoftwareSerial.h>
SoftwareSerial mySerial(8,9); // RX, TX
mySerial.begin(115200);

matthijskooijman, 2017-11-03:
I just tried this on Arduino IDE 1.8.5-ish (some version from git), compiling for the Arduino Uno Wifi as well, which works. I do get a ton of warnings that I should probably fix (including the "has a base '...' whose type uses the anonymous namespace" warnings you're seeing), so I guess those are not related to the linker errors you're seeing.

Could you create another log with "verbose compilation" enabled in the Arduino IDE preferences? I just want to check if the "fields.cpp" file, which is supposed to define these variables.

Just a hunch, could you also try changing [this line](https://github.com/matthijskooijman/arduino-dsmr/blob/master/src/dsmr/fields.cpp#L42) to:

    constexpr char dsmr::fields::identification::name_progmem[];

And then see if the linker error related to this variable disappears? It shouldn't matter, but just in case resolution of this variable doesn't use the "using namespace" directives above it properly.

IncognitoCore, 2017-11-03:
I have modified the line you mentioned and enabled verbose logging for verifying and uploading.

I'm using Arduino IDE "Arduino 1.8.5" and Board "Arduino UNO WiFi".

[Output when verifiying.](https://www.dropbox.com/s/80llnf7vyx0337q/error_messages_2.txt?dl=0)

I've put the dsmr library in my Documents folder (Mac). Is that correct? This has always worked for other libraries I used.

```
 kevin > ~/Documents/Arduino/libraries/dsmr > tree
 .
 ├── dsmr
 │   ├── fields.cpp
 │   ├── fields.h
 │   ├── parser.h
 │   ├── reader.h
 │   └── util.h
 └── dsmr.h
 
 1 directory, 6 files
```

[These](https://www.dropbox.com/sh/q2a4l1mxmwj74hd/AACR52Sxk69IWKXfY0iXpEDIa?dl=0) are the files that I'm using.

matthijskooijman, 2017-11-03:
Ah, you removed the "src" folder, which is needed for the Arduino IDE to do recursive compilation of source files (so it's completely not compiling the cpp file now). You should just download the entire library as it is (including the library.properties file as well), or just make a git clone.

IncognitoCore, 2017-11-03:
I moved it to my Documents/Arduino folder because the git clone on this repo didnt work for me as is. Then it complains about not finding dsmr.h:

`/Users/kevin/Developer/arduino-dsmr/examples/read/read.ino:12:18: fatal error: dsmr.h: No such file or directory
 #include "dsmr.h"
                  ^
compilation terminated.
exit status 1
Error compiling for board Arduino Uno WiFi.
`

matthijskooijman, 2017-11-03:
You should have something that looks like this:

```
 kevin > ~/Documents/Arduino/libraries/dsmr > tree
 .
 ├── src
 │   ├── dsmr.h
 │   └── dsmr
 │       ├── fields.cpp
 │       ├── fields.h
 │       ├── parser.h
 │       ├── reader.h
 │       └── util.h
 ├── README.md
 ├── examples
 └── library.properties
```

matthijskooijman, 2017-11-03:
(which is what you get if you git clone into the `libraries` dir)

IncognitoCore, 2017-11-03:
I had cloned the repo in `~/Development/arduino-dsmr`. When I clone the repo into `~/Documents/Arduino/libraries/dsmr` it works fine indeed.
Thanks for the support!

matthijskooijman, 2017-11-03:
Good to hear it's working now. I'm closing this issue, then.

closed 2017-11-03 by matthijskooijman