matthijskooijman/arduino-dsmr #8: util/crc16.h not found

mrWheel, 2017-12-11:
Hi Matthijs,

I want to use this library but I get a compile error:

```
Arduino: 1.8.3 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

../Documents/com~apple~CloudDocs/ArduinoProjects/libraries/arduino-dsmr-master/examples/parse/parse.ino:12:
../Documents/com~apple~CloudDocs/ArduinoProjects/libraries/arduino-dsmr-master/src/dsmr/parser.h:34:24: fatal error: util/crc16.h: No such file or directory
 #include <util/crc16.h>
                        ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

Can you point me in the right direction on how to obtain this crc16 library??

matthijskooijman, 2017-12-11:
This should be fixed by the PR I just merged (which includes a couple of fixes that should make things work on ESP). Your issue made me realize I still had to review the latest changes to the PR and looked now, so I merged it.

Could you test and report if this fixes your problem?

closed 2017-12-11 by matthijskooijman

mrWheel, 2017-12-12:
Hi Matthijs,

matthijskooijman, 2017-12-12:
The actual commit was create a while ago, I just merged the pull-request (containing that commit) yesterday. So I created a "merge commit", which doesn't actually change any files, it just pulls in the older commit.

Does that make things clearer?

mrWheel, 2017-12-12:
Eh.. no, not really :-( 
But thats probably due to my ignorance.

Just don't understand how "not changing any files" makes this repository work.
But it does and now I will try to make a "Slimme Meter 2 WiFi bridge"!

Regards

matthijskooijman, 2017-12-12:
Well, I guess it's just a peculiarity of how git or github shows "last changed" times. It shows the time when a change was first made (on the developers computer), not the time when it actually changed / merged in this particular repository.

mrWheel, 2017-12-12:
Ok, that makes sense!

Again: hartstikke bedankt!

commit 45221d4a (2017-12-11): Merge pull request #6 from ewasscher/master

Multiple changes for building on non-AVR platforms

This fixes #8.