matthijskooijman/arduino-dsmr #9: Not a real issue, just a call for explanation

mrWheel, 2017-12-12:
Matthijs,

I can see this is a very neat and smart written library, but the whole concept of "templates" is far beyond my "C++" knowledge.

Can (will) you please explain (preferable with a short piece of code) how I can obtain certain values?

For example:
How can I obtain the value of "timestamp" or "energy_delivered_tariff1"?

Thanks!

matthijskooijman, 2017-12-12:
Did you check out the README? It has some examples, though it is a bit limited. I'll repeat the most relevant pieces here:

> As an example, consider we want to parse the identification and current power fields in the example message above. We define a datatype:

```
using MyData = ParsedData<
  /* String */ identification,
  /* FixedValue */ power_delivered
>;
```

> The syntax is a bit weird because of the template magic used, but the above essentially defines a struct with two members. There is some extra stuff in the background, but the MyData can be used just like the below struct. It also takes up the same amount of space.
```
struct MyData {
	String identification;
	FixedValue power_delivered;
};
```

In other words, you can just access the members directly:

    pdel1 += data.energy_delivered_tariff1;
    Serial.println(data.timestamp);

Note that the type of the members varies. In particular there is a "FixedValue" type (used for `energy_delivered_tariff1` among others). The README has some info on it, but you can usually just use it as if it were a float, as should happen in my example above.

Let me know if this works for you!

mrWheel, 2017-12-12:
Matthijs,

I will try!

But first I have an other, bigger problem.

I'm trying to read the input with EspSoftwareSerial but that does not work...

My testbed is a processor that sends the raw data array from parse.ino every 10 seconds. I have changed the read.ino script with the following code:

```
#include "dsmr.h"
#include <EspSoftwareSerial.h>       // <== new
.
#define SERIAL_RX     D5             //  pin for SoftwareSerial RX
#define INVERTED      false          // true in real live
SoftwareSerial  SerialS(SERIAL_RX, -1, INVERTED); // (RX, TX. inverted, buffer)
.
P1Reader reader(&SerialS, 0);        // <== changed
.

void setup() {
  Serial.begin(115200);
  SerialS.begin(115200);             // <== changed
  #ifdef VCC_ENABLE
    // This is needed on Pinoccio Scout boards to enable the 3V3 pin.
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
  #endif

  // start a read right away
  reader.enable(true);
  last = millis();
}
```
But nothings shows!

So I added this code in the beginning of "loop()":
```
void loop () {
  while(SerialS.available()) {
    Serial.print((char)SerialS.read());
  }
.
.
```
.. and then I see the exact data send from the other computer. I even checked if the '\r' and '\n' characters are in the right place .. and they do!! 

So, what am I doing wrong??

Sorry for being such a pain.
Hope you can help!

matthijskooijman, 2017-12-12:
I got a comment notification with another question, but I can't find it here now. Did you fix the problem and delete your comment? :-)

mrWheel, 2017-12-12:
Hi Matthijs,

Yes, I had a problem using EspSoftwareSerial as input for DSMR. It seems that the library cannot cope with 115200 baud. It worked on lower baud-rates.

matthijskooijman, 2018-09-18:
This seems resolved, so I'm closing this issue.

closed 2018-09-18 by matthijskooijman