matthijskooijman/arduino-dsmr #10: Example with SoftwareSerial

zuidwijk, 2018-04-07:
I'm trying to figuring out how to use your library with SoftwareSerial. I have configured it like this:

```
#include "dsmr.h"
#include <SoftwareSerial.h>
#define SERIAL_RX D5 // pin for SoftwareSerial RX
SoftwareSerial mySerial(SERIAL_RX, -1, true); 
```

And tried each following:

```
P1Reader mySerial;
P1Reader reader(mySerial);
P1Reader reader(&mySerial);
```

Is it posible to put an example sketch in your source for how to configure this?

matthijskooijman, 2018-04-08:
I recall that I tried SS as well, but ended up building a hardware inverter and using hardware serial. IIRC SS was not entirely reliable, but that might have been only when combined with some of my other code, it's a long time ago. IIRC it should at least give some results.

Looking at your code suggestions, I think none of those would compile, so I guess you're stuck in figuring out how to write this down? Looking at the [P1Reader constructor](https://github.com/matthijskooijman/arduino-dsmr/blob/master/src/dsmr/reader.h#L72) you need to pass a `Stream*` and an int for the request pin, so that would be:

P1Reader reader(&mySerial, 2);

(replace 2 with whatever pin you've connected to the request pin). See also https://github.com/matthijskooijman/arduino-dsmr/blob/master/examples/read/read.ino#L115

zuidwijk, 2018-04-10:
Got it! 

```
#include <SoftwareSerial.h>
#define SERIAL_RX D5 // pin for SoftwareSerial RX
#define MAXLINELENGTH 64
SoftwareSerial mySerial(SERIAL_RX, -1, true, MAXLINELENGTH); // (RX, TX. inverted, buffer)
P1Reader reader(&mySerial, 2);
```

The request pin will give 3.3v, is that enough? Or should it be fixed with a transistor switching 5v?

matthijskooijman, 2018-04-10:
Dunno off-hand, the DSMR spec might know. Perhaps just try and see?

matthijskooijman, 2018-09-18:
I believe this issue can be closed? If not, feel free to add more comments.

closed 2018-09-18 by matthijskooijman

Sieberkev, 2020-01-09:
> I recall that I tried SS as well, but ended up building a hardware inverter and using hardware serial. IIRC SS was not entirely reliable, but that might have been only when combined with some of my other code, it's a long time ago. IIRC it should at least give some results.

Hello @matthijskooijman 

I'm very sorry to hijack this thread, but can you perhaps post the inverter you used to replace the SoftwareSerial? I tried with a 2N2222 transistor and a buttload of different resistor values found on random places on the internet, but I can not get a reversed signal and I'm about to start flipping tables at this point :expressionless:

Thanks in advance :blush:

matthijskooijman, 2020-01-23:
Here's a schematic I had lying around. I never got around to actually using this in practice, but I think the schematic matches what I tested on a breadboard back then:

[image]

The P1 connector is on the top, the serial signal on the right. On the left, I have two transistors to build a high-side 5V switch for the request pin, which I needed since my Arduino (rather, a Pinoccio board) was 3.3V.