matthijskooijman/arduino-dsmr #11: Just what I needed!

timkoers, 2018-04-20:
Hero!

matthijskooijman, 2018-04-20:
Thanks :-)

What do you mean by baudrate increase?

timkoers, 2018-04-21:
I'm trying to read out ma Iskra MT174 with an IR Head. The default connection is at 300bps and when sending the right aknowlegdment the baudrate switches to 9600 bps.
I already found out that this library is nothing more than a parser, so I'll check to see if I can edit this library so that I can make it work 🙂

matthijskooijman, 2018-04-21:
Ah, I only used it on a P1 port at a fixed baudrate, didn't know there was a baudrate switching thingy :-)

The library contains a small reader class, which IIRC takes care of the request pin and reading from a serial port. I guess it could be adapted (or a similar thing could be added) to control the baudrate for your usecase. One tricky thing there is that there is no virtual base class for all serial implementations, so you'll probably need a template class to make it work for e.g. both HardwareSerial and SoftwareSerial.

CapitalFH, 2018-10-27:
Is it possible to use this project with ESMR5 (new smartmeter) this meter has a other parity on p1 8n1 instead of 7e1 ?

matthijskooijman, 2019-05-04:
@timkoers, did you ever make this work?

matthijskooijman, 2020-03-16:
For lack of response, I'm closing this. If this is still relevant, feel free to comment or reopen.

closed 2020-03-16 by matthijskooijman