matthijskooijman/arduino-dsmr #12: parser not working with ISKRA (AM550)

mrWheel, 2018-09-18:
Hi Matthijs,

I build a wireless logger  with your great library and everything works great..

But now I'm trying to use a working logger with a different Smart Meter and everything goed haywire!

It works with this meter **Landis Gyr+ (SMR 5.0)**

but not with this meter **ISKRA (AM550)**
With the ISKRA only part of the telegram is parsed correct (the Energy Delivered/Returned records) but the Power records give strange results. But the parser does not report any errors(?).

Have you heard of this problem before and is there a fix for it?

If you need more information, just let me know!

Regards

mrWheel, 2018-09-18:
Matthijs,

What happens if a field is not used (send) by this meter, but I still process it "as if it is a valid field"?
Is the value initiated to zero or has it a random value?

To elaborate:

If the Smart Meter only has 1 fase but I do process the other 2 fases

```
     current_l2
     current_l3
```

If I now process current_l1 I will get a valid value for the current. But what if I process ```current_l2```? Is the value of this field "what ever was in memory" or has it the value "0" (zero)?

If it is random, is there a way to initialise the "```MyData```" array?

Sorry for this stupid question, but the whole Template thing is way over my head!!!

mrWheel, 2018-09-18:
Matthijs,

It is indeed a initialisation problem!

Can you please tell me how I can initialise all MyData fileds before I parse de telegram received?
Depending on the meter at hand some data is in the telegram and some are not. But I don't want to change the program (and webpage) for every type of situation (one fase, three fases etc.). So in my setup it is absolutely necessaire to initialise all fields to zero so, if they are not in the telegram at least they have a zero value... 

Thanks in advance!

matthijskooijman, 2018-09-18:
It's been a while, but I think you can just call `current_l2.present()` to find out if the field was present in the parsed message. This works, because the fields are not simple fields (e.g. not just a `String`, `FixedValue`, etc.) but a struct that can be implicitly converted to the underlying type, but which also has a `present()` method. See https://github.com/matthijskooijman/arduino-dsmr/blob/45221d4afaf8e5d7eb8ac9ae764b2ab4596833a2/src/dsmr/fields.h#L170-L179

matthijskooijman, 2018-09-18:
Ok, scratch that, while trying to improve the README, I found that things are a bit different from what I remember. Apparently the `present()` method is overridden for each field, so you can only use that when you convert to the right type. Instead, it seems you can just use the `current_l2_present` field defined in line 173 instead.

Let me know if that works for you?

mrWheel, 2018-09-18:
YES! That works!

Thanks again for your prompt reply!

matthijskooijman, 2018-09-18:
I've improved the README to explicitly mention these `xxx_present` fields. I think this issue is now resolved. If you have further suggestions on how the docs could be improved in this area, feel free to leave more comments.

closed 2018-09-18 by matthijskooijman