matthijskooijman/arduino-dsmr #14: Print raw-format of a telegram

mrWheel, 2018-09-19:
How do I use this : 

```
    /**
     * Returns the data read so far.
     */
    const String &raw() {
      return buffer;
    }
```
If it should do what I thing it would be very nice to be able to print the raw format of the telegram!

matthijskooijman, 2018-09-19:
IIRC, you should be able to do:
```
  if (reader.available()) {
    Serial.println("Message:");
    Serial.println(reader.raw());
    MyData data;
    String err;
    if (reader.parse(&data, &err)) {
      // Parse succesful, print result
      data.applyEach(Printer());
    } else {
      // Parser error, print error
      Serial.println(err);
    }
  }
```

This is taken from the read.ino example, but with the first two `println`s added. IOW, `raw()` returns the data read so far, but once you call `parse()`, the buffer is cleared again. So if you retrieve the buffer when `available()` is true, but before calling `parse()`, you should get the complete message. Note that the buffer does not include the leading `/` and the trailing checksum.

matthijskooijman, 2019-05-04:
@mrWheel, could you confirm if my suggestion worked?

mrWheel, 2019-05-05:
I can confirm that this code does work:
```
    if (reader.available()) {
        MyData data;
        String err;
        if (reader.parse(&data, &err)) {  // Parse succesful, print result
          processData(data);
          data.applyEach(showValues());
          printData();
        } else {                                    // Parser error, print error
          Serial.printf("Parse error %s\r\n", err.c_str());
        }
    }
```

matthijskooijman, 2019-05-05:
@mrWheel thanks for your response. HOwever, AFAICS your code does not print the raw, unparsed message which was the subject of this issue :-)

I would want to close this issue, but ideally by adding the raw printing to the read.ino example. But I only want that after making sure it actually works (and I do not have a DSMR-attached setup currently to test with, so I was hoping you could confirm).

mrWheel, 2019-05-05:
Ok, I will test your code tomorrow