matthijskooijman/arduino-dsmr #16: incomplete execution of the code

Daniel-Nabil-Artil, 2018-09-29:
your code compiles but when comes to execution it stops at the line:
 if (reader.available()) {
which means no reading happens and no data received! 

would appreciate your insight what could the problem be? my hardware is Arduino Uno with inverted RX and TX through using : 
SoftwareSerial mySerial(8,9,1);
and a 10k resistor to boost the data voltage.
So don't know if the problem is in the hardware or somewhere else!

matthijskooijman, 2018-09-29:
I would suggest a simple sketch that reads from the `mySerial` port and prints to `Serial`, to see if there is any data incoming at all.

If no data comes in, you should check your software serial setup, or perhaps use a scope or logic analyzer to see if any data is sent on the wire (you can [build one from a Uno](https://github.com/gillham/logic_analyzer) if you don't have one).

If data comes in, the data might end up being invalid. You might be able to spot this in the output, already. IIRC the reader detects this from an invalid CRC, but does not currently report such an error (would be a good addition to the library, though). You could check this by adding some Serial prints in this area (probably in the else of that if): https://github.com/matthijskooijman/arduino-dsmr/blob/f79c9061d558b65a4877045e7867e93af025e736/src/dsmr/reader.h#L136-L144

Daniel-Nabil-Artil, 2018-09-29:
I did try a simpler code and recieved decoded readings ... well the recieved messages were sometimes disrupted but I recieved them anyway with this : 
`SoftwareSerial mySerial(8,9,1);
char telegram;
 
void setup() {
    Serial.begin(115200);
    //altSerial.begin(115200);
    mySerial.begin(115200);
}
 

void loop() {
    if (mySerial.available()) {
        telegram = mySerial.read();
   
        // --- 7 bits instelling ---
        telegram &= ~(1 << 7);
        char inChar = (char)telegram;

    Serial.print(telegram);
    
    }
}`
so no idea why don't I recieve anything with your code!

matthijskooijman, 2018-09-29:
Ok, so then it is likely that there is just an error in every message, which is detected using the CRC and causes the entire message to be ignored.

One other thought: You remove the upper bit from each character in the sketch you showed. Is that really necessary to get valid data (iow, if you remove the `telegram &= ~(1 << 7);`, do you no longer get valid messages)? If so, then the Dsmr library might also need to do this masking for reception to work, since I do not think it currently applies this mask.

Daniel-Nabil-Artil, 2018-10-03:
I have been trying since and no readings!
what did you mean by "seems there are still occasional reception errors when using SoftwareSerial." as I am using the SoftwareSerial to do the inversion ... how did you do it?

matthijskooijman, 2018-10-03:
> what did you mean by "seems there are still occasional reception errors when using SoftwareSerial."

When using SS, I still noticed that every now and then a character would be received incorrectly (e.g. one bit would be flipped, probably). If one character in a message is wrong, the checksum will fail and the entire message will be rejected (since the library cannot tell *which* character is wrong).

>  as I am using the SoftwareSerial to do the inversion ... how did you do it?

In the end, I switched to using hardware serial with a hardware-based inverter. However, I never actually started to use that solution, I'm not actually using my own library yet :-p