matthijskooijman/arduino-dsmr #17: incorrect parsing of identification and electricity_failure_log

robertoostenveld, 2018-10-06:
My ISKRA AM550 sends messages like this according to Serial.println(reader.raw())

```
ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(181006151740S)
0-0:96.1.1(4530303433303037313330393836393138)
1-0:1.8.1(000045.342*kWh)
1-0:1.8.2(000088.857*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.252*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00008)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00005)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(238.8*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.258*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383134313338343138)
0-1:24.2.1(181006151505S)(00032.834*m3)
```

The first telegram is parsed correctly like this

```
identification: ISK5\2M550E-1012
p1_version: 50
timestamp: 181006151740S
equipment_id: 4530303433303037313330393836393138
energy_delivered_tariff1: 45.34kWh
energy_delivered_tariff2: 88.86kWh
energy_returned_tariff1: 0.00kWh
energy_returned_tariff2: 0.00kWh
electricity_tariff: 0001
power_delivered: 0.25kW
power_returned: 0.00kW
electricity_failures: 8
electricity_long_failures: 2
electricity_failure_log: ()
electricity_sags_l1: 5
electricity_swells_l1: 1
message_long: 
voltage_l1: 238.80V
current_l1: 1A
power_delivered_l1: 0.26kW
power_returned_l1: 0.00kW
gas_device_type: 3
gas_equipment_id: 4730303339303031383134313338343138
gas_delivered: 32.83m3
```

The subsequent telegrams result in "Duplicate field" parse errors pointing to the very first ISK5\2M550E-1012 line. To work around this problem I commended out line 114 and 115 in parser.h. After that the parsing works (most of the times, sometimes there is some obvious junk in the message). However, I see that the identification field is growing in subsequent telegrams like this:

```
identification: ISK5\2M550E-1012
identification: ISK5\2M550E-1012ISK5\2M550E-1012
identification: ISK5\2M550E-1012ISK5\2M550E-1012ISK5\2M550E-1012
```

The same happens for 

```
electricity_failure_log: ()
electricity_failure_log: ()()
electricity_failure_log: ()()()
```

I suspect that this is might be the underlying cause of the  "Duplicate field" parse errors, since it seems that the previous values are not cleared. 

My C++ template skills are not good enough to figure out why this would happen.  Right now my workaround is to ignore the parse errors by keeping line 114 and 155 commented out and by not including identification and electricity_failure_log in the ParsedData template.

matthijskooijman, 2018-10-06:
Can you post your sketch? I suspect you're putting the parsed message in a global variable perhaps, so its contents are not cleared between runs (I'm not actually sure if there's a way to clear a parsed message right now, haven't checked the code).

robertoostenveld, 2018-10-07:
Thanks for your reply.  The sketch is [esp8266_p1_thingspeak.ino](https://github.com/robertoostenveld/arduino/blob/master/esp8266_p1_thingspeak/esp8266_p1_thingspeak.ino) from my [arduino repository](
https://github.com/robertoostenveld/arduino). The "MyData data" object/structure(?) is indeed global, which I could change. 

It would be nice if your dsmr library would have a method to clear it, but feel free to close this issue if you are not planning to implement it.

matthijskooijman, 2018-10-07:
I had a closer look at the code now, and I think the default constructors are available. I believe this means you should be able to write:

    data = MyData();

to construct a new data object and assign it. You might even do:

    data = {};

to call the default constructor without having to name the type, but I'm not sure if that's valid syntax.

That would probably solve your problem for now. I do wonder if this should perhaps happen automatically whenever `parse()` is called? OTOH, that might be inefficient when it is not needed (e.g. when passing a fresh data object every time)...