matthijskooijman/arduino-dsmr #19: Gasmeter values not available

jicho, 2018-12-02:
Hi,

Since yesterday I'm the happy owner of the Slimme Meter uitlezer (https://opencircuit.nl/Blog/Slimme-meter-uitlezer). It works perfectly for electricity, but for some reason the gas values (usage and equipment id) of my telegram are rendered to 0 m3 and and unknown equipment ID.

Willem, the creator of the Slimme Meter uitlezer, advised me to make changes to your lib or contact the creator of the DSMR parser lib. Maybe my meter is sending other info than other meters... Since I have no clue where to start looking I hope that the brains behind this lib has an idea why my gas meter values are not detected :)

My telegram look lik this:
```
/ISK5\2M550E-1012                                                               
                                                                                
1-3:0.2.8(50)                                                                   
0-0:1.0.0(181202194148W)                                                        
0-0:96.1.1(453030***************9323137)                                  
1-0:1.8.1(000617.195*kWh)                                                       
1-0:1.8.2(000730.512*kWh)                                                       
1-0:2.8.1(000000.000*kWh)                                                       
1-0:2.8.2(000000.000*kWh)                                                       
0-0:96.14.0(0001)                                                               
1-0:1.7.0(00.192*kW)                                                            
1-0:2.7.0(00.000*kW)                                                            
0-0:96.7.21(00009)                                                              
0-0:96.7.9(00004)                                                               
1-0:99.97.0(2)(0-0:96.7.19)(180116115413W)(0000002996*s)(181103104217W)(00000042
71*s)                                                                           
1-0:32.32.0(00009)                                                              
1-0:32.36.0(00001)                                                              
0-0:96.13.0()                                                                   
1-0:32.7.0(225.3*V)                                                             
1-0:31.7.0(000*A)                                                               
1-0:21.7.0(00.193*kW)                                                           
1-0:22.7.0(00.000*kW)                                                           
0-2:24.1.0(003)                                                                                                                                       
0-2:96.1.0(47303033**************31333137)                                  
0-2:24.2.1(181202194007W)(00283.687*m3)                                         
!9F30    
```

Do you have any idea why I get an empty value for the equipment id and the gas usage?

Looking forward to your reaction.

WJ4IoT, 2018-12-02:
It takes time for the gas meter is connected to the main meter; do you see any readings of gas on the electricity meter itself?

jicho, 2018-12-03:
Hi, yes I see gas readings on the electricity meter. And with the USB cable connected to the meter for DSMR reader (a python telegram parser) I had gas readings. 

When I understand the telegram correctly the telegram contains gas reading:

```
0-2:24.1.0(003)                                                                                                                                       
0-2:96.1.0(47303033**************31333137)                                  
0-2:24.2.1(181202194007W)(00283.687*m3)                                         
```

What I see in other examples is that these lines start with 0-1, mine start with 0-2... Maybe because it is my second meter? Just guessing ;)

jongdesteven, 2018-12-03:
I have the same with my setup; they replaced my gas meter because it was not working, the new gas meter registred itself with ID 2. It is then parsed as a water meter; no difference whatsoever

> What I see in other examples is that these lines start with 0-1, mine start with 0-2... Maybe because it is my second meter? Just guessing ;)

jicho, 2018-12-03:
I don’t think your water meter assumption is incorrect... according to this document:
https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf

You can have a gasmeter installed on different channels:
```
7.3 Gas Data
The following is only applicable if Gas meters are connected. The OBIS reference Channel number (n) will be defined by the installing order of the M-Bus devices. So a Gas meter can be installed on channel 1, 2, 3 or 4.
```

Same with the water meter... I get the feeling that the logic needs to figure out which device is on which channel. 

Lucky for us (not) the replacement meter was registered on the second channel... For what I see that it okay according to the specs, but kind of strange from a counting perspective ;)

jongdesteven, 2018-12-03:
I meant that this software is assuming the device on channel 2 is a water meter.
When you look in the specification, there is no difference in the data for a gas meter and water meter. Therefor this software cannot distinguish betweent the two.

jicho, 2018-12-03:
Ah okay, so... the designers of the DSMR standard is the place to “complain” and the code works just fine :)

Weird that they just skipped such a definition for additional devices...

matthijskooijman, 2018-12-03:
I'm a bit late to the party, but this was sortof documented at https://github.com/matthijskooijman/arduino-dsmr#slave-meters I guess it would be nicer to have some way to config this without changing the library itself, though...

matthijskooijman, 2018-12-03:
@jicho, I think it's best to keep this issue closed, but it seems good to track this as a feature request in a new issue. Care to open it? I think there can be two parts to the request: Either improve documentation or error reporting about slave meters with non-standard ids, and making it easier to support non-standard ids.

jicho, 2018-12-03:
@matthijskooijman: done, look at this https://github.com/matthijskooijman/arduino-dsmr/issues/20

matthijskooijman, 2018-12-03:
Perfect, thanks!