matthijskooijman/arduino-dsmr #23: 'file structure' not working

jcjveraa, 2019-11-08:
Thanks for making this!

I spent about an hour staring at messages from the compiler similar to 

> undefined reference to `dsmr::fields::timestamp::id'

Mostly caused by my inexperience I guess. It seems that for (my installation?) of the Arduino compiler to recognize your code, I need to have dsmr.h and all other files from the src directory in the same directory (and editing dsmr.h to 'not refer to the dsmr directory')

Or is there some other way to make this work?

matthijskooijman, 2019-11-08:
I'm not sure what is happening for you exactly, but this library is intended to be installed as an Arduino library in the `libraries` subdirectory of your sketchbook. Essentially download [the zip](https://github.com/matthijskooijman/arduino-dsmr/archive/master.zip) and unpack it into that `libraries` dir. See https://www.arduino.cc/en/guide/libraries for more info.

jcjveraa, 2019-11-08:
Ah, I just cloned the repository and just included the header. Thanks for
pointing this out :)

matthijskooijman, 2019-11-08:
You can also clone the repository directly into your `libraries` directly, that should also work :-)

jcjveraa, 2019-11-09:
Was probably missing that library file then - I only copied the src folder
to my libraries (from the 'repos' folder where all my github files go).

Thanks, have a good weekend!

matthijskooijman, 2019-11-09:
So I assume you've solved this? Then I'll go ahead and close the issue. Good weekend to you to! :-)

closed 2019-11-09 by matthijskooijman