matthijskooijman/arduino-dsmr #24: Need help with lab experiment

arjenv, 2019-11-15:
Hi Matthijs
Thank you for this library. I am having problems getting it to work at the labtable. I installed the DSMRlogger program from Aandewiel and that parses data. However I am trying to get a labtable setup with a PIC controller emulating the DSMR smart meter, and I do not get any result, not even an error..... The Aandewiel program lets me see the raw data from the smart meter and -when connected to the PIC controller- indeed outputs the sent data. However it is not parsed, nothing, not even an error, just nothing. 

The PIC sends out a telegram. Unlike the smart meter it throws out the data in one blow, no pauses in between. Every line is ended with a 0x0D, 0x0AD:

/KFM5KAIFA-METER                                                                
                                                                                
1-3:0.2.8(42)                                                                   
0-0:1.0.0(190805204420S)                                                        
0-0:96.1.1(4530303236303030303237313334363135)                                  
1-0:1.8.1(004029.456*kWh)                                                       
1-0:1.8.2(002611.355*kWh)                                                       
1-0:2.8.1(000000.000*kWh)                                                       
1-0:2.8.2(000000.000*kWh)                                                       
0-0:96.14.0(0002)                                                               
1-0:1.7.0(00.163*kW)                                                            
1-0:2.7.0(00.000*kW)                                                            
0-0:96.7.21(00012)                                                              
0-0:96.7.9(00005)                                                               
1-0:99.97.0(6)(0-0:96.7.19)(190508111811S)(0000008169*s)(190406211046S)(0000011)
1-0:32.32.0(00001)                                                              
1-0:52.32.0(00003)                                                              
1-0:72.32.0(00004)                                                              
1-0:32.36.0(00000)                                                              
1-0:52.36.0(00000)                                                              
1-0:72.36.0(00000)                                                              
0-0:96.13.1()                                                                   
0-0:96.13.0()                                                                   
1-0:31.7.0(000*A)                                                               
1-0:51.7.0(000*A)                                                               
1-0:71.7.0(000*A)                                                               
1-0:21.7.0(00.157*kW)                                                           
1-0:41.7.0(00.000*kW)                                                           
1-0:61.7.0(00.011*kW)                                                           
1-0:22.7.0(00.000*kW)                                                           
1-0:42.7.0(00.000*kW)                                                           
1-0:62.7.0(00.000*kW)                                                           
0-1:24.1.0(003)                                                                 
0-1:96.1.0(4730303332353631323533373030343135)                                  
0-1:24.2.1(190805200000S)(03293.167*m3)                                         
!5A99 

I changed some of your example program to get this:

matthijskooijman, 2019-11-15:
I just had a look at your code. If you get now output at all (not even the "now here"), then that suggests the reader never flags data as available. That could happen when it reads wrong checksums, are you sure you're calculating them correctly? Are you including the 0xD, 0xA in the checksum (From reading my own back, I think you should)?

You could add some debug output around this line to see if the checksum is the problem: https://github.com/matthijskooijman/arduino-dsmr/blob/a3d9113da5a094cea7040a7c98f5de44a3aba0cd/src/dsmr/reader.h#L136

arjenv, 2019-11-16:
If your code does not response whenever the checksum is wrong, I strongly believe that that is the case in my lab experiment. Can I somehow simply omit the CRC check in your code? I´d like to change some values randomly in the PIC without having to calculate the checksum every time.

I´l try to put some debug statements around your suggested lines, however I am not a C++ programmer.

arjenv, 2019-11-16:
Debug statements around the code you suggest (as in ´Serial.print´) Throws an exception(28). Pls show me how I can deactivate the check for CRC.

matthijskooijman, 2019-11-16:
"An exception(28)" is a bit vague, not sure what that is exactly (AFAIK Arduino compiles with C++ exceptions disabled, but maybe this is something else).

As for the disabling CRC, the actual check happens in the if I linked you: https://github.com/matthijskooijman/arduino-dsmr/blob/a3d9113da5a094cea7040a7c98f5de44a3aba0cd/src/dsmr/reader.h#L136

Even with no C++ experience, I expect that is readable to you? It checks if parsing the CRC was succesful and then if the CRC value read matches the value calculated. To disable the check, simply replace that line with `if (true) {`.

arjenv, 2019-11-24:
the lines are readable, however I do not have oversight unless I could read the entire program. So I thought as much and tried your suggestion and that seems to do the trick for my needs. Thanks again. I will close it as an issue (cause it is not).

closed 2019-11-24 by arjenv