matthijskooijman/arduino-dsmr #30: Bug in parsing logic

Mindavi, 2020-01-28:
Hi, I want to bring a bug to your attention. I found this while fuzzing the library. The issue appears when an invalid input with more than 3 'decimals' is given.

The issue is that the max_decimals value is post-decremented, which means that, if the max_decimals value hits 0, the loop stops, but the max_decimals value is then wrapped around to the max uint value. This will trigger a very long process that will decrement the max_decimals value back to 0.

An example is shown below. Note that, for fuzzing purposes, I've disabled checksum checking in the build that generated this. To provoke the actual behavior, you might need to update the checksum/disable checksum checking.

<details>
<summary>example 1</summary>

```
/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(200112213629W)
0-0:96.1.1(3330300333833330303332323639333333)
1-0:1.8.1(.00385.6
00900)
0-1*m3)
!B1C3
```

</details>

<details>
<summary>example 2</summary>

```
/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(200112213629W)
0-0:96.1.1(3330300333833330303332323639333333)
1-0:1.8.1(0.0385.6
00900)
0-1*m3)
!B1C3
```

</details>

By changing line 201 to the following, it can be fixed.
https://github.com/matthijskooijman/arduino-dsmr/blob/a3d9113da5a094cea7040a7c98f5de44a3aba0cd/src/dsmr/parser.h#L201

```patch
-      while(num_end < end && !strchr("*)", *num_end) && max_decimals--) {
+      while(num_end < end && !strchr("*)", *num_end) && max_decimals) {
+        --max_decimals;
```

Because I wanted to fuzz on a x86-64 target, I've also updated the code to support running on linux. This might have helped triggering this bug, but I'm not sure about that.

I could share the adapted version, but it's mostly the same with basically all `String` usage changed with `std::string`.

Mindavi, 2020-02-27:
FYI: I've since published the fork.
https://github.com/Mindavi/arduino-dsmr

matthijskooijman, 2022-01-02:
Fixed by #34.

closed 2022-01-02 by matthijskooijman