matthijskooijman/arduino-dsmr #37: Problem compiling demo code

erhapp, 2020-09-06:
Hello

I tried to compile the demo code and got the following error:

`.../.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/minimal_parse.ino.cpp.o:(.text._ZN4dsmr10ParsedDataIINS_6fields14identificationENS1_15power_deliveredEEE10parse_lineERKNS_6ObisIdEPKcS9_[dsmr::ParsedData<dsmr::fields::identification, dsmr::fields::power_delivered>::parse_line(dsmr::ObisId const&, char const*, char const*)]+0x4): undefined reference to `dsmr::fields::identification::id'`

Am I doing something wrong? If so, can you please point me in the right direction?

matthijskooijman, 2020-09-06:
How did you install the library? Sounds a bit like you added the header files into your sketch, but the cpp files might not be compiled?

erhapp, 2020-09-06:
Indeed, stupid me. It appears some files didn't get copied to the library directory. Thank you very much for taking the time to point this out. Everything works fine now.

closed 2020-09-06 by erhapp

matthijskooijman, 2020-09-06:
Well, let's say I recognized the problem because you're not the first to run into it :-)

Good luck!