matthijskooijman/arduino-dsmr #38: reader.available() remains false using Wemos d1 mini

ErwinP, 2021-02-09:
Hi,

I'm trying to use this library in combination with a Wemos d1 mini v3 and hardwareserial. I live in Belgium, and got a sagemcom t211 installed. I'm aware that Belgium uses a slightly different OBIS syntax, but this is not the problem.

I have managed to connect and get a read by applying [this ]https://camo.githubusercontent.com/158c00212c0317c87e4b58bbb55741fd1aaba2635b23497b06765c4950b92da5/68747470733a2f2f77696b692e7069657065722e65752f77702d636f6e74656e742f75706c6f6164732f323031362f30332f5031706f72742e706e67) scheme from [silvanverschuur ](https://github.com/silvanverschuur/esp-p1-mqtt). However, given the several benefits from using your library, I'd like to start using your library. I started off by using the `read.ino` from the examples folder.

I have tried a lot of setups to connect the Wemos to the P1 port, but can't seem to get past the `if(reader.available())` statement: it never becomes `true`. My gut feeling tells me that it is due to the lack of the request pin, but I can't seem to find out how to properly connect it. The most simple solution, making a direct connection from P1's pin 3 to Wemos pin D2, isn't doing the trick. 

I don't know how to proceed from this point on, and my stubbornness is already causing quite some sleep deprivation. Any help on this matter would be incredibly appreciated.

Thank you in advance,
Erwin

matthijskooijman, 2021-02-09:
IIRC checksum errors can also prevent the reader from returning `available()` as `true` (but I also think that error logging here isn't ideal currently).

If there is no request pin, does the port always send unconditionally?

It would be good to confirm if serial is receiving any bytes at all, maybe by adding some debug prints in the library.

See also #27 about differences between the Belgian and Dutch DSMR formats. IIRC the Belgian format is not really supported, but as you suggest, I would just expect missing fields rather than no data at all (though maybe failed parsing prevents available, not entirely sure).

keukenrol, 2023-02-27:
Hi Erwin, do you have your setup running? I have an esp32 and a transistor to invert the data from the T211 to the rx2 input. However when using the read example I dont get actual data. I know the output does work correctly when uploading my own sketch and just showing what is read from serial2 and printing it on serial.

SplinterII, 2023-05-19:
From the DSMR specs I understand the Data request (input) line expects 5V. The pins of the Wemos deliver 3.3, which is below the treshold from the specs. So I made the data request line permanently high by connecting it to the Wemos 5V pin. Then all works as expected....