matthijskooijman/arduino-dsmr #39: ESP 3.0.0

MarcFinns, 2021-06-15:
Your lib works like a charm, i have been using it for a while.
Today I made a slight modification to my sketch and I realised that it does not compile anymore with the newest ESP package.
I worked around the issue by downgrading, but I thought to notify you...

matthijskooijman, 2021-06-15:
Thanks for reaching out. Any particular error message or more detailed problem description you could share?

MarcFinns, 2021-06-16:
Hi Matthijs, fair question...
I re-upgraded to 3.0.0 and compiled your parse.ino example, which works fine with ESP 2.7.4.
See below the errors I see - with verbose flag enabled. Thanks again, keep up the great work!

==============================
Arduino: 1.8.13 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 4MB (FS:1MB OTA:~1019KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 3000000"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Marc/Library/Arduino15/packages -hardware /Users/Marc/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Marc/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Marc/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=4M1M,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=3000000 -ide-version=10813 -build-path /Users/Marc/Documents/Arduino/builds -warnings=default -build-cache /var/folders/my/_tr269dj3j3003ny56h2tync0000gp/T/arduino_cache_528829 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.0-newlib4.0.0-gnu23-48f7b08.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs-3.0.0-newlib4.0.0-gnu23-48f7b08.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.0-newlib4.0.0-gnu23-48f7b08 -verbose /Users/Marc/Documents/Arduino/libraries/arduino-dsmr-master/examples/parse/parse.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Marc/Library/Arduino15/packages -hardware /Users/Marc/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Marc/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Marc/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=4M1M,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=3000000 -ide-version=10813 -build-path /Users/Marc/Documents/Arduino/builds -warnings=default -build-cache /var/folders/my/_tr269dj3j3003ny56h2tync0000gp/T/arduino_cache_528829 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.0-newlib4.0.0-gnu23-48f7b08.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs-3.0.0-newlib4.0.0-gnu23-48f7b08.path=/Users/Marc/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.0-newlib4.0.0-gnu23-48f7b08 -verbose /Users/Marc/Documents/Arduino/libraries/arduino-dsmr-master/examples/parse/parse.ino
Using board 'generic' from platform in folder: /Users/Marc/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0
Using core 'esp8266' from platform in folder: /Users/Marc/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0
Detecting libraries used...
/Users/Marc/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/Marc/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/tools/sdk/include -I/Users/Marc/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/tools/sdk/lwip2/include -I/Users/Marc/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/tools/sdk/libc/xtensa-lx106-elf/include -I/Users/Marc/Documents/Arduino/builds/core -c -w -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -D [... cut ...]
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/Marc/Documents/Arduino/libraries/arduino-dsmr-master/src/dsmr/fields.h:346:1: note: in expansion of macro 'DEFINE_FIELD'
  346 | DEFINE_FIELD(slave_delivered, TimestampedFixedValue, ObisId(0, SLAVE_MBUS_ID, 24, 2, 1), TimestampedFixedField, units::m3, units::dm3);
      | ^~~~~~~~~~~~
Using library arduino-dsmr-master at version 0.1 in folder: /Users/Marc/Documents/Arduino/libraries/arduino-dsmr-master 
exit status 1
Error compiling for board Generic ESP8266 Module.

matthijskooijman, 2021-06-16:
Thanks! I edited your comment to put triple backticks (\```) around your paste, so it shows as a code block, making it a bit more readable.

Looks like you're seeing the same issue already reported at #36, which I still have to fix... That issue has a workaround (though you'll have to puzzle it together from the comments a bit...).

matthijskooijman, 2021-06-16:
I'm closing this as duplicate now, if you have more comments, best add them in the other issue.

closed 2021-06-16 by matthijskooijman

MarcFinns, 2021-06-16:
great, i did not notice #36. my bad. thanks again

matthijskooijman, 2021-06-16:
No problem!

MarcFinns, 2021-06-16:
just in case / FYI. 
Small box, self powered, OTA and WiFi setup included. Powered by your lib!
[https://github.com/MarcFinns/ESP-P1-Meter](url)