matthijskooijman/arduino-dsmr #41: Hoog cpu gebruik

jimmyd-be, 2021-08-02:
Sinds kort is mijn Raspberry pi 4 extreem traag. Na wat onderzoek lijkt het erop dat de python en gunicorn van dsmr reader bijna 100% van de cpu opgebruikt.

Ik gebruik de remote datalogger en op de server zelf de laatste docker stack. De sever zelf is waar het allemaal traag loopt. De remote datalogger lijkt niets aan de hand.

Kan hier iets aan gedaan worden?

matthijskooijman, 2021-08-02:
@jimmyd-be, weet je zeker dat je hier aan het goede adres bent? Deze repository gaat over een library voor Arduino, op microcontrollers, geschreven in C++, en heeft voor zover ik kan zien geen relatie met een python-programma op een rpi (tenzij iemand deze code op de een of andere manier in een pythonproces heeft verwerkt, maar dan kan ik daar niet echt ondersteuning op bieden...).

jimmyd-be, 2021-08-02:
My bad. Dit is idd de verkeerde Repo

closed 2021-08-02 by jimmyd-be