matthijskooijman/arduino-dsmr #43: Cannot assign  to String with latest Arduino after fix for Error: a reinterpret_cast is not a constant expression

Roos-AID, 2021-12-27:
I have upgraded my Arduino environment and EPS8266 libraries.  I did run in the Error: a reinterpret_cast is not a constant expression and applied the fix a mentioned. 

With the old version I was able to compile and run all my code.  Since this upgrade I get error 
read:98:12: error: conversion from 'const dsmr::fields::NameConverter<dsmr::fields::identification>' to non-scalar type 'String' requested
   98 |     String Name = Item::name;

I have been able to reduce the code in read.ino example to produce the error. I am lost, dont know how to fix this , I hope you can help. 

This is the code in read.ino that fails during compile. Again this did work before the reinterpret_cast problem.
`struct Printer {
  template<typename Item>
  void apply(Item &i) {
    String Name = Item::name;
    // DO WHAT WE WANT WITH String ..... 
  }
};
`

Detailed error during compile :
`/Users/xxx/Documents/Arduino/read/read.ino: In instantiation of 'void Printer::apply(Item&) [with Item = dsmr::fields::identification]':
/Users/xxx/Documents/Arduino/libraries/arduino-dsmr-master/src/dsmr/fields.h:46:12:   required from 'void dsmr::ParsedField<T>::apply(F&) [with F = Printer; T = dsmr::fields::identification]'
/Users/xxx/Documents/Arduino/libraries/arduino-dsmr-master/src/dsmr/parser.h:129:13:   required from 'void dsmr::ParsedData<T, Ts ...>::applyEach_inlined(F&&) [with F = Printer&; T = dsmr::fields::identification; Ts = {dsmr::fields::p1_version, dsmr::fields::timestamp, dsmr::fields::equipment_id, dsmr::fields::energy_delivered_tariff1, dsmr::fields::energy_delivered_tariff2, dsmr::fields::energy_returned_tariff1, dsmr::fields::energy_returned_tariff2, dsmr::fields::electricity_tariff, dsmr::fields::power_delivered, dsmr::fields::power_returned, dsmr::fields::electricity_threshold, dsmr::fields::electricity_switch_position, dsmr::fields::electricity_failures, dsmr::fields::electricity_long_failures, dsmr::fields::electricity_failure_log, dsmr::fields::electricity_sags_l1, dsmr::fields::electricity_sags_l2, dsmr::fields::electricity_sags_l3, dsmr::fields::electricity_swells_l1, dsmr::fields::electricity_swells_l2, dsmr::fields::electricity_swells_l3, dsmr::fields::message_short, dsmr::fields::message_long, dsmr::fields::voltage_l1, dsmr::fields::voltage_l2, dsmr::fields::voltage_l3, dsmr::fields::current_l1, dsmr::fields::current_l2, dsmr::fields::current_l3, dsmr::fields::power_delivered_l1, dsmr::fields::power_delivered_l2, dsmr::fields::power_delivered_l3, dsmr::fields::power_returned_l1, dsmr::fields::power_returned_l2, dsmr::fields::power_returned_l3, dsmr::fields::gas_device_type, dsmr::fields::gas_equipment_id, dsmr::fields::gas_valve_position, dsmr::fields::gas_delivered, dsmr::fields::thermal_device_type, dsmr::fields::thermal_equipment_id, dsmr::fields::thermal_valve_position, dsmr::fields::thermal_delivered, dsmr::fields::water_device_type, dsmr::fields::water_equipment_id, dsmr::fields::water_valve_position, dsmr::fields::water_delivered, dsmr::fields::slave_device_type, dsmr::fields::slave_equipment_id, dsmr::fields::slave_valve_position, dsmr::fields::slave_delivered}]'
/Users/xxx/Documents/Arduino/libraries/arduino-dsmr-master/src/dsmr/parser.h:124:22:   required from 'void dsmr::ParsedData<T, Ts ...>::applyEach(F&&) [with F = Printer; T = dsmr::fields::identification; Ts = {dsmr::fields::p1_version, dsmr::fields::timestamp, dsmr::fields::equipment_id, dsmr::fields::energy_delivered_tariff1, dsmr::fields::energy_delivered_tariff2, dsmr::fields::energy_returned_tariff1, dsmr::fields::energy_returned_tariff2, dsmr::fields::electricity_tariff, dsmr::fields::power_delivered, dsmr::fields::power_returned, dsmr::fields::electricity_threshold, dsmr::fields::electricity_switch_position, dsmr::fields::electricity_failures, dsmr::fields::electricity_long_failures, dsmr::fields::electricity_failure_log, dsmr::fields::electricity_sags_l1, dsmr::fields::electricity_sags_l2, dsmr::fields::electricity_sags_l3, dsmr::fields::electricity_swells_l1, dsmr::fields::electricity_swells_l2, dsmr::fields::electricity_swells_l3, dsmr::fields::message_short, dsmr::fields::message_long, dsmr::fields::voltage_l1, dsmr::fields::voltage_l2, dsmr::fields::voltage_l3, dsmr::fields::current_l1, dsmr::fields::current_l2, dsmr::fields::current_l3, dsmr::fields::power_delivered_l1, dsmr::fields::power_delivered_l2, dsmr::fields::power_delivered_l3, dsmr::fields::power_returned_l1, dsmr::fields::power_returned_l2, dsmr::fields::power_returned_l3, dsmr::fields::gas_device_type, dsmr::fields::gas_equipment_id, dsmr::fields::gas_valve_position, dsmr::fields::gas_delivered, dsmr::fields::thermal_device_type, dsmr::fields::thermal_equipment_id, dsmr::fields::thermal_valve_position, dsmr::fields::thermal_delivered, dsmr::fields::water_device_type, dsmr::fields::water_equipment_id, dsmr::fields::water_valve_position, dsmr::fields::water_delivered, dsmr::fields::slave_device_type, dsmr::fields::slave_equipment_id, dsmr::fields::slave_valve_position, dsmr::fields::slave_delivered}]'
/Users/xxx/Documents/Arduino/read/read.ino:207:31:   required from here
read:98:12: error: conversion from 'const dsmr::fields::NameConverter<dsmr::fields::identification>' to non-scalar type 'String' requested
   98 |     String Name = Item::name;
      |            ^~~~`

Hope you can point me in right direction, 
Regards,
Rob

matthijskooijman, 2022-01-02:
The problem here is that `Item::name` is now a NameConverter object, rather than a `const __FlashStringHelper*` as before, and that C++ does not allow two user-defined conversions in a single expression (NameConverter to __FlashStringHelper operator *and* __FlashStringHelper to String using a constructor).

I tried fixing this by adding a conversion from NameConverter to String directly, but that breaks `Serial.println(Item::name)` again (since that is no ambiguous, it can either convert to __FlashStringHelper or String and both can be printed).

What does work now, is adding an explicit conversion to String, e.g.:

```
String name = (String)Item::name;
```

Or, with the latest master version (which includes a fix for the reinterpret_cast issue), the recommended approach is to use `get_name()` instead of `name`, which should work exactly like `name` before the fix:
```
String name = Item::get_name();
```

matthijskooijman, 2022-01-02:
I think this should solve your issue. If not, let me know and we can reopen it.

closed 2022-01-02 by matthijskooijman

Roos-AID, 2022-01-03:
Mathijs,

thanks for the fast response and fix. It seems to work,  it compiles without issues for ESP8266, but I dont have a spare ESP8266 to test at the moment.  Don’t want to flash on my working ESP8266 as I have  to convert SPIFFS to LittleFS.

I was planning to go to ESP32, unfortunatly GCC has caused other problems as well in the ESP32 setup, so need some more work to get everything in my code ESP32 compatible, which is quite a struggle…. but I will manage.  

Met vriendelijk groet,

Rob Roos

e-mail : ***@***.***

*****DISCLAIMER*****
De informatie in deze e-mail is vertrouwelijk en uitsluitend bestemd voor de geadresseerde.
Indien de lezer van deze mededeling niet de geadresseerde is, wordt U er hierbij op gewezen,
dat U geen recht hebt kennis te nemen van de rest van deze e-mail, het te kopiëren of te verstrekken aan andere personen dan de geadresseerde.
Indien U deze e-mail abusievelijk hebt ontvangen, brengt U dan de afzender op de hoogte waarbij U gevraagd zal worden het originele bericht te vernietigen.
Verzender is niet verantwoordelijk en wijst iedere aansprakelijkheid af voor en/of in verband met alle gevolgen en/of schade van een juiste en volledige verzending en ontvangst van de inhoud en van deze e-mail!
Door elektronische verzending kunnen aan de inhoud van dit bericht géén rechten worden ontleend!