matthijskooijman/arduino-dsmr #46: Update for PlatformIO

thomasvnl, 2022-12-21:
Hi Matthijs,

Could you please update the corresponding repo on PlatformIO for this library? I'm getting the reinterpret cast issue that you've fixed with a previous update when using PlatformIO, like with ESPHome.

Thank you in advance and with kind regards,
Thomas

matthijskooijman, 2022-12-21:
I'm not sure I can - I never registered anything with PIO and do not really use it myself. Do you know how PIO updates work? Maybe I just need to bump the version and/or create a tag maybe?

SteynGuelen, 2023-01-24:
@thomasvnl you can add:
lib_deps = 
    https://github.com/matthijskooijman/arduino-dsmr.git
    
  to your platformio.ini file, then it gets the latest version from github.