matthijskooijman/arduino-dsmr #47: Baud rate check in header parsing too strict

Hadramal, 2023-02-07:
Hello!

I bought a Slimmelezer+ from Marcel Zuidwijk and it seems to use this parser in the firmware built upon ESPHome. It connects to the electricity meter just fine but can't read the data:

`19:04:32 | [E] | [dsmr:265] | /ADN9 6534  
^ 
Invalid identification string`

Now this seems to be because of this line in this library:

`// The first identification line looks like:
// XXX5<id string>
// The DSMR spec is vague on details, but in 62056-21, the X's
// are a three-leter (registerd) manufacturer ID, the id
// string is up to 16 chars of arbitrary characters and the
// '5' is a baud rate indication. 5 apparently means 9600,
// which DSMR 3.x and below used. It seems that DSMR 2.x
// passed '3' here (which is mandatory for "mode D"
// communication according to 62956-21), so we also allow
// that.
if (line_start + 3 >= line_end || (line_start[3] != '5' && line_start[3] != '3'))
return res.fail(F("Invalid identification string"), line_start);`

I.e. unless the fourth char (after the /, which is read before) is 3 or 5, the lib stops reading data. But, that fourth char is the baud rate according to IEC62056-21 (and your comment) and could be many things. In Sweden the standard is 115200 which is the number 9. My Aidon meter spits out data like this:
`/ADN9 6534

0-0:1.0.0(213112235959W)
1-0:1.8.0(12345678.123*kWh)
1-0:2.8.0(12345678.123*kWh)
1-0:3.8.0(12345678.123*kVarh) 
....
1-0:71.7.0(123.1*A)! `

It seems like this check is far too strict? Baud rate can be left to ESPHome or whatever other device uses this lib? It looks like without this header verification the device could parse "my" format as well, and 9 seems to be a valid number according to the spec.

matthijskooijman, 2023-02-08:
Thanks for your report. It seems the check is indeed too strict, so I just removed the check entirely in git master.

Note that I'm not sure if esphome uses my library directly, I suspect there might be some forks floating around, so it might not start working immediately for you.

Also, could you maybe paste a complete telegraf from your meter here? I would like to collect some examples of different formats for future automated testing :-)

closed 2023-02-08 by matthijskooijman

Hadramal, 2023-02-08:
First of all thank you!

Yes - after making the issue I realised ESPHome is using a fork of your
code, so I will raise an issue with the maintainers of that repo as well...
I could have made a pullrequest myself but I'm not sure I fully understand
the ESPHome build system so I'm not sure how I could test it! I normally
write Java so everything's familiar yet strange.

As the reader I have stops reading data after the header I can't give you
actual live data, but I got the example from this document:
https://www.skekraft.se/wp-content/uploads/2022/10/SKEKRAFT1-1026973-v2-Aidon_HAN_Interface_RJ12.pdf
 - it has examples from their meters in the back of the pdf. It's in
Swedish though, but hopefully you won't miss the data examples! The numbers
refer to their different power meters, with 6534 being the far most common
in residential homes. As they were early here in Sweden they also on their
early installs often send the data in a HDLC frame, examples of that are
also in the pdf, but that is being phased out

/Henrik

matthijskooijman, 2023-02-15:
Thanks! I've added the spec to the repo for future reference :-)

marcus-aa, 2023-03-02:
Resolved, but here are some complete telegrams from a swedish AIdon 6534.
`/ADN9 6534

0-0:1.0.0(230302093810W)
1-0:1.8.0(00031157.405*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000857.557*kVArh)
1-0:4.8.0(00000918.106*kVArh)
1-0:1.7.0(0004.344*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.314*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0001.083*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0001.855*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0001.408*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.351*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:44.7.0(0000.545*kVAr)
1-0:63.7.0(0000.507*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(231.9*V)
1-0:52.7.0(232.6*V)
1-0:72.7.0(232.7*V)
1-0:31.7.0(004.9*A)
1-0:51.7.0(008.3*A)
1-0:71.7.0(006.4*A)
!7D79
/ADN9 6534

0-0:1.0.0(230302093820W)
1-0:1.8.0(00031157.405*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000857.557*kVArh)
1-0:4.8.0(00000918.106*kVArh)
1-0:1.7.0(0004.320*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.313*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0001.063*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0001.853*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0001.411*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.329*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:44.7.0(0000.545*kVAr)
1-0:63.7.0(0000.505*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(231.7*V)
1-0:52.7.0(232.4*V)
1-0:72.7.0(232.8*V)
1-0:31.7.0(004.8*A)
1-0:51.7.0(008.3*A)
1-0:71.7.0(006.4*A)
!8D80
/ADN9 6534

0-0:1.0.0(230302093830W)
1-0:1.8.0(00031157.405*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000857.557*kVArh)
1-0:4.8.0(00000918.106*kVArh)
1-0:1.7.0(0005.984*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.332*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0001.057*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0002.688*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0002.233*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.356*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:44.7.0(0000.960*kVAr)
1-0:63.7.0(0000.963*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(231.8*V)
1-0:52.7.0(231.8*V)
1-0:72.7.0(231.5*V)
1-0:31.7.0(004.8*A)
1-0:51.7.0(012.3*A)
1-0:71.7.0(010.4*A)
!806A
/ADN9 6534

0-0:1.0.0(230302093840W)
1-0:1.8.0(00031157.405*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000857.557*kVArh)
1-0:4.8.0(00000918.106*kVArh)
1-0:1.7.0(0004.327*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.302*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0001.058*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0001.859*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0001.410*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.347*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:44.7.0(0000.540*kVAr)
1-0:63.7.0(0000.508*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(232.1*V)
1-0:52.7.0(232.9*V)
1-0:72.7.0(232.9*V)
1-0:31.7.0(004.8*A)
1-0:51.7.0(008.3*A)
1-0:71.7.0(006.4*A)
!120C
/ADN9 6534

0-0:1.0.0(230302093850W)
1-0:1.8.0(00031157.405*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000857.557*kVArh)
1-0:4.8.0(00000918.106*kVArh)
1-0:1.7.0(0004.319*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.309*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0001.055*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0001.856*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0001.409*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.331*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:44.7.0(0000.541*kVAr)
1-0:63.7.0(0000.507*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(232.1*V)
1-0:52.7.0(232.7*V)
1-0:72.7.0(232.8*V)
1-0:31.7.0(004.7*A)
1-0:51.7.0(008.3*A)
1-0:71.7.0(006.4*A)
!1C56`

commit ad239375 (2023-02-08): Remove checks on the identification string contents

It turns out some meters (Swedish, apparently) use a different baud rate
indicator number. But rather than adding that number, this just removes
all checking on the identification string and just passes it on
unmodified, for maximum flexibility with future meters.

This fixes #47.