matthijskooijman/arduino-dsmr #50: T211 DSMR5 meter does not send a unit for unlinked gas meter, causing 'missing unit' errors

jeffreykog, 2023-04-25:
This issue was reported upstream at https://github.com/esphome/issues/issues/3886. The T211 meter is used by Enexis on 3-phase service. When no gas meter is present, or when it is not linked yet by the service provider, the meter reports a gas reading without m3 unit. This causes the library to throw a 'missing unit' error, which should probably be handled more gracefully.

Example: 
```
0-1:24.2.1(632525252525S)(00000.000)
          ^
Missing unit
```

The ID of the gas meter in this example line also seems to be a placeholder. Possibly this identifier (`632525252525S`) can be used to detect a gas meter that is not properly linked yet, allowing an empty value without unit to be returned.

Example correct line:
```
0-1:24.2.1(xxxx25070000S)(14336.324*m3)
```

This is where the error originates: https://github.com/glmnet/arduino-dsmr/blob/master/src/dsmr/parser.h#L220

SamuraiCoder, 2024-02-21:
Just had my meter replaced and got exactly the same error. I was able to make the rest work by commenting in the yaml about the gas but I am wondering when I will be able to read the gas since its was interests me more to monitor. 

Also, EnergieDirect / Enexis need to activate the remote reading I guess cause still I am not reading anything on their website.

Bascy, 2025-03-01:
We modified our fork of dsmr to allow missing units when the value is 0.
Yo can find it here https://github.com/I-Connect/arduino-dsmr