matthijskooijman/arduino-dsmr #52: arduino-dsmr niet werkt op nieuwe esp8266 lib

tthandroid, 2023-10-29:
Dag Matthijs,
Heb je een suggestie voor me om de arduino-dsmr
draaien te maken voor de laatst arduino-esp8266 lib?
Alvast bedankt.

Mvg
tthandroid

PolarGoose, 2023-10-30:
Hello @tthandroid,

Could you provide more details?
I'm using the **arduino-dsmr** lib from Arduino IDE and it has been working so far. I guess that it uses the latest **arduino-esp8266**.

tthandroid, 2023-10-30:
Hi Matthijs,, 

When I switched to the " https://github.com/esp8266/Arduino" Release 3.1.2 my ESP8266 web server hanged om reading the p1 data.
I don't have this issue in de release 1.0.0..

If you sure that the arduino-dsmr lib will wokrs on de last /esp8266/Arduino last release, than I'm going to search further.

Thanks for your answer!

Regards,
ttandroid.

PolarGoose, 2023-10-30:
@tthandroid,

> When I switched to the " https://github.com/esp8266/Arduino" Release 3.1.2 my ESP8266 web server hanged om reading the p1 data.

There is an Arduino sketch [arduino-dsmr/blob/master/examples/read/read.ino](https://github.com/matthijskooijman/arduino-dsmr/blob/master/examples/read/read.ino). Does it work for you if you try to use it on your board? This will allow you to check if it is actually **arduino-dsmr** lib and not something else,

PolarGoose, 2023-10-30:
@tthandroid,

The `read.ino` prints the output into the debug USB-serial port, you need to check what output you get to see if `arduino-dsmr` works.

tthandroid, 2023-10-30:
Hi Matthijs,

Thank you so far for your time!
I have used the ESP01-S with the schematic provided in Tasmota Gitbub builded the prototype for the interface. I'm not able to do the test.
I have to make a new prototype with two serial ports, so that it will provide me the possibility to do the test.

Regards,