matthijskooijman/arduino-dsmr #54: Unable to build for ESP32 Dev Module

Park0, 2023-11-13:
I can't build minimal_parse in de arduino 1.8.19 IDE but also platformio gives the same error. For now i have no clue how to fix this.

```
error: 'identification' was not declared in this scope
```

PolarGoose, 2023-11-15:
Hello Park0,
I have just double checked and the [minimal_parse](https://github.com/matthijskooijman/arduino-dsmr/tree/master/examples/minimal_parse) example compiles without problems:
[image]
Could you provide a complete build log?

Park0, 2023-11-15:
Install esp32 2.0.6
```
minimal_parse:28:16: error: 'identification' was not declared in this scope
   /* String */ identification,
                ^~~~~~~~~~~~~~
minimal_parse:29:20: error: 'power_delivered' was not declared in this scope
   /* FixedValue */ power_delivered
                    ^~~~~~~~~~~~~~~
minimal_parse:30:1: error: template argument 1 is invalid
 >;
 ^
minimal_parse:30:1: error: template argument 2 is invalid
Documents\Arduino\libraries\arduino-dsmr\examples\minimal_parse\minimal_parse.ino: In function 'void setup()':
minimal_parse:35:3: error: 'MyData' was not declared in this scope
   MyData data;
   ^~~~~~
minimal_parse:36:44: error: 'data' was not declared in this scope
   ParseResult<void> res = P1Parser::parse(&data, msg, lengthof(msg));
                                            ^~~~
Documents\Arduino\libraries\arduino-dsmr\examples\minimal_parse\minimal_parse.ino:36:44: note: suggested alternative: 'atan'
   ParseResult<void> res = P1Parser::parse(&data, msg, lengthof(msg));
                                            ^~~~
                                            atan
```

PolarGoose, 2023-11-16:
@Park0,
Could you contact me via my email [goosepolar@gmail.com](https://github.com/matthijskooijman/arduino-dsmr/pull/goosepolar@gmail.com)? we can arrange a call, and I can take a look at what is going on.